auto-pi-lot/autopilot #17: Add Autopilot info to open-neuroscience.com

amchagas, 2020-07-10:
Hello Autopilot folks!

We are reaching out because we would love to have your project listed on [Open Neuroscience](<https://open-neuroscience.com>), and also share information about this project:

Open Neuroscience is a community run project, where we are curating and highlighting open source projects related to neurosciences!

Briefly, we have a website where short descritptions about projects are listed, with links to the projects themselves, their authors, together with images and other links. 

Once a new entry is made, we make a quick check for spam, and publish it. 

Once published, we make people aware of the new entry by [Twitter](https://twitter.com/openneurosci) and a [Facebook](https://www.facebook.com/OpenNeuroscience) group. 

**To add information about their project, developers only need to fill out this [form](https://forms.gle/ByM8thAhZJkHBMQN8)**

In the form, people can add subfields and tags to their entries, so that projects are filterable and searchable on the website!

The reason why we have the form system is that it makes it open for everyone to contribute to the website and allows developers themselves to describe their projects! 

Also, there are so many amazing projects coming out in Neurosciences that it would be impossible for us to keep track and log them all!

Please [get in touch](mailto:openeuroscience@gmail.com) if you have any questions or would like to collaborate!

sneakers-the-rat, 2020-08-04:
Absolutely! Thanks for getting in touch. I'm just about to push a major new version, so i'll submit then :)

sneakers-the-rat, 2020-12-14:
wow time elapsed and this didn't get done. reopening

sneakers-the-rat, 2021-07-19:
Finally did this! https://open-neuroscience.com/post/autopilot/

closed 2021-07-19 by sneakers-the-rat