auto-pi-lot/autopilot #19: Missing directories in install

sneakers-the-rat, 2020-09-16:
Submitted in discussion board:
https://groups.google.com/g/autopilot-users/c/HUCdRJJ-Sg4/m/ag3rwzYABQAJ

>  Minor issues I observed where that I ran (in both the “pure” pip install and the pip install from the repo) into the error of a missing dir under ‘./autopilot/external/pigpio’ which could be solved by simply creating the dir. Also, git-cloning the ‘wehr-lab’ repo missed the contents for both folders in the src which I added by hand. Afterwards the installation was successful and I ran the setup script also without errors (not caring about disabling bluetooth for now)

cxrodgers, 2020-09-24:
Thanks for these hints. They helped me get going too. I'm using a brand-new raspberry pi 4B with latest version of raspbian. I had a few additional issues, which may be related to using a newer pi or just that I haven't had any previous versions installed.

First I had to install some dependencies manually. They're listed at [the bottom of the installation documentation](https://docs.auto-pi-lot.com/en/latest/guide.installation.html#requirements), but I didn't notice this until later.

`sudo apt install build-essential  cmake git  python3-dev  libatlas-base-dev  libsamplerate0-dev  libsndfile1-dev  libreadline-dev libasound-dev  i2c-tools  libportmidi-dev  liblo-dev  libhdf5-dev  libzmq-dev libffi-dev libdb-dev libsystemd-dev
`
The last two were not in the documentation but I needed them (or maybe they would have been installed by something else).

Here's how I cloned everything to get around the missing links issue. I cloned into a directory called `~/dev/autopilot`, not sure if the doc assumes you clone into `~/autopilot`?

```
mkdir ~/dev
cd ~/dev
git clone https://github.com/wehr-lab/autopilot
git clone https://github.com/sneakers-the-rat/pigpio autopilot/src/pigpio
git clone https://github.com/sneakers-the-rat/mlx90640-library autopilot/src/mlx90640-library
```

Now to install:
```
cd autopilot # this line is missing from installation webpage
pip3 install --user -e .
```

Ok so that should work, except what I actually had to do is `sudo pip3 install -e .`. I could not install both editable and user at the same time, I think it's related to this recent issue in pip:
https://github.com/pypa/pip/issues/7953

I don't like to `pip install` with `sudo`, so I might try to get the fix on that issue page working before proceeding.

sneakers-the-rat, 2020-09-29:
OK this should be fixed in v0.3.2, the pr linked above.

- clarified the documentation -- there are some system dependencies that need to be installed before the autopilot configuration is run, so I just put them all up front (and also made an option to install them in the configuration menus)
- the expectation of those directories was a hangover from a previous way i had intended to do packaging, so removed that. pigpio should be installed from requirements now. 

thanks for your notes on documentation, really appreciated.

permissions on the pi can be weird -- i can never tell if you need to `--user` out of an installation or not. Yes you should use `--user` if you dont want an editable install, and i would encourage using a venv if you want an editable install :). 

i'll close for now, but reopen if issue persists.

closed 2020-09-29 by sneakers-the-rat