auto-pi-lot/autopilot #23: Setup Error

roaldarbol, 2020-10-12:
I'm using a RasPi 4B with Raspbian Buster 10 and Python 3.7. I've managed to successfully install autopilot and all dependencies, but when running:
``
python3 -m autopilot.setup.setup_autopilot
``
I keep getting the following error:

``
/usr/lib/python3.7/runpy.py:125: RuntimeWarning: 'autopilot.setup.setup_autopilot' found in sys.modules after import of package 'autopilot.setup', but prior to execution of 'autopilot.setup.setup_autopilot'; this may result in unpredictable behaviour
  warn(RuntimeWarning(msg))
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/pi/.local/lib/python3.7/site-packages/autopilot/setup/setup_autopilot.py", line 691, in <module>
    setup.run()
  File "/home/pi/.local/lib/python3.7/site-packages/npyscreen/apNPSApplication.py", line 30, in run
    return npyssafewrapper.wrapper(self.__remove_argument_call_main)
  File "/home/pi/.local/lib/python3.7/site-packages/npyscreen/npyssafewrapper.py", line 41, in wrapper
    wrapper_no_fork(call_function)
  File "/home/pi/.local/lib/python3.7/site-packages/npyscreen/npyssafewrapper.py", line 97, in wrapper_no_fork
    return_code = call_function(_SCREEN)    
  File "/home/pi/.local/lib/python3.7/site-packages/npyscreen/apNPSApplication.py", line 25, in __remove_argument_call_main
    return self.main()
  File "/home/pi/.local/lib/python3.7/site-packages/npyscreen/apNPSApplicationManaged.py", line 148, in main
    self.onStart()
  File "/home/pi/.local/lib/python3.7/site-packages/autopilot/setup/setup_autopilot.py", line 535, in onStart
    self.pilot_1 = self.addForm('CONFIG_PILOT_1', Pilot_Config_Form_1, name="Setup Pilot Agent - 1/2")
  File "/home/pi/.local/lib/python3.7/site-packages/npyscreen/apNPSApplicationManaged.py", line 55, in addForm
    fm = FormClass( parentApp=self, *args, **keywords)
  File "/home/pi/.local/lib/python3.7/site-packages/autopilot/setup/setup_autopilot.py", line 228, in __init__
    super(Autopilot_Form, self).__init__(*args, **kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/npyscreen/fmForm.py", line 70, in __init__
    self.create()
  File "/home/pi/.local/lib/python3.7/site-packages/autopilot/setup/setup_autopilot.py", line 498, in create
    self.populate_form(PILOT_PREFS)
  File "/home/pi/.local/lib/python3.7/site-packages/autopilot/setup/setup_autopilot.py", line 270, in populate_form
    widget = self.add(nps.TitleText, name=param['text'], value = str(default))
  File "/home/pi/.local/lib/python3.7/site-packages/npyscreen/fmForm.py", line 384, in add_widget
    *args, **keywords)
  File "/home/pi/.local/lib/python3.7/site-packages/npyscreen/wgtitlefield.py", line 69, in __init__
    self.make_contained_widgets()
  File "/home/pi/.local/lib/python3.7/site-packages/npyscreen/wgtitlefield.py", line 86, in make_contained_widgets
    self.label_widget = textbox.Textfield(self.parent, relx=self.relx, rely=self.rely, width=len(self.name)+1, value=self.name, color=self.labelColor)
  File "/home/pi/.local/lib/python3.7/site-packages/npyscreen/wgtextbox.py", line 44, in __init__
    self.update()
  File "/home/pi/.local/lib/python3.7/site-packages/npyscreen/wgtextbox.py", line 132, in update
    self._print()
  File "/home/pi/.local/lib/python3.7/site-packages/npyscreen/wgtextbox.py", line 299, in _print
    color
_curses.error: addwstr() returned ERR
``

Any ideas why this appears?

roaldarbol, 2020-10-12:
Also, at random times, this variation occurs instead:

``/usr/lib/python3.7/runpy.py:125: RuntimeWarning: 'autopilot.setup.setup_autopilot' found in sys.modules after import of package 'autopilot.setup', but prior to execution of 'autopilot.setup.setup_autopilot'; this may result in unpredictable behaviour``
``  warn(RuntimeWarning(msg))``
``Traceback (most recent call last):``
``  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main``
``    "__main__", mod_spec)``
``  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code``
``    exec(code, run_globals)``
``  File "/home/pi/.local/lib/python3.7/site-packages/autopilot/setup/setup_autopilot.py", line 691, in <module>``
``    setup.run()``
``  File "/home/pi/.local/lib/python3.7/site-packages/npyscreen/apNPSApplication.py", line 30, in run``
``    return npyssafewrapper.wrapper(self.__remove_argument_call_main)``
``  File "/home/pi/.local/lib/python3.7/site-packages/npyscreen/npyssafewrapper.py", line 41, in wrapper``
``    wrapper_no_fork(call_function)``
``  File "/home/pi/.local/lib/python3.7/site-packages/npyscreen/npyssafewrapper.py", line 97, in wrapper_no_fork``
``    return_code = call_function(_SCREEN)    ``
``  File "/home/pi/.local/lib/python3.7/site-packages/npyscreen/apNPSApplication.py", line 25, in __remove_argument_call_main``
``    return self.main()``
``  File "/home/pi/.local/lib/python3.7/site-packages/npyscreen/apNPSApplicationManaged.py", line 148, in main``
``    self.onStart()``
``  File "/home/pi/.local/lib/python3.7/site-packages/autopilot/setup/setup_autopilot.py", line 535, in onStart``
``    self.pilot_1 = self.addForm('CONFIG_PILOT_1', Pilot_Config_Form_1, name="Setup Pilot Agent - 1/2")``
``  File "/home/pi/.local/lib/python3.7/site-packages/npyscreen/apNPSApplicationManaged.py", line 55, in addForm``
``    fm = FormClass( parentApp=self, *args, **keywords)``
``  File "/home/pi/.local/lib/python3.7/site-packages/autopilot/setup/setup_autopilot.py", line 228, in __init__``
``    super(Autopilot_Form, self).__init__(*args, **kwargs)``
``  File "/home/pi/.local/lib/python3.7/site-packages/npyscreen/fmForm.py", line 70, in __init__``
``    self.create()``
``  File "/home/pi/.local/lib/python3.7/site-packages/autopilot/setup/setup_autopilot.py", line 498, in create``
``    self.populate_form(PILOT_PREFS)``
``  File "/home/pi/.local/lib/python3.7/site-packages/autopilot/setup/setup_autopilot.py", line 270, in populate_form``
``    widget = self.add(nps.TitleText, name=param['text'], value = str(default))``
``  File "/home/pi/.local/lib/python3.7/site-packages/npyscreen/fmForm.py", line 384, in add_widget``
``    *args, **keywords)``
``  File "/home/pi/.local/lib/python3.7/site-packages/npyscreen/wgtitlefield.py", line 69, in __init__``
``    self.make_contained_widgets()``
``  File "/home/pi/.local/lib/python3.7/site-packages/npyscreen/wgtitlefield.py", line 105, in make_contained_widgets``
``    **self._passon)``
``  File "/home/pi/.local/lib/python3.7/site-packages/npyscreen/wgtextbox.py", line 21, in __init__``
``    super(TextfieldBase, self).__init__(screen, **keywords)``
``  File "/home/pi/.local/lib/python3.7/site-packages/npyscreen/wgwidget.py", line 271, in __init__``
``    self.set_size()``
``  File "/home/pi/.local/lib/python3.7/site-packages/npyscreen/wgwidget.py", line 414, in set_size``
``    raise NotEnoughSpaceForWidget("Not enough space: max y and x = %s , %s. Height and Width = %s , %s " % (my, mx, self.height, self.width) ) # unsafe. Need to add error here.``
``npyscreen.wgwidget.NotEnoughSpaceForWidget: Not enough space: max y and x = 0 , 131. Height and Width = RAISEERROR , 131 ``

roaldarbol, 2020-10-12:
OK, I've solved the issue - but definitely a bug that needs a fix. The problem was that my terminal was not wide enough: If I stretched it far enough before running ``python3 -m autopilot.setup.setup_autopilot`` everything works just fine, but if my terminal is narrow I get the errors.

roaldarbol, 2020-10-13:
I just installed on my Macbook with the same error and same solution.

sneakers-the-rat, 2020-10-14:
Yes -- have come across this before. need to document & give more informative error message, or ideally make the setup routine more flexible to not need to use the cli gui.

sneakers-the-rat, 2020-10-17:
Resolved by providing an informative error message, this error now presents like this:

[image]

https://github.com/wehr-lab/autopilot/commit/acf734a3db5ac0fddd6265ec89e6944b98aec33c
https://github.com/wehr-lab/autopilot/commit/d819388cfe2b3b5f8b7cce303975d2410ce64b4f

will integrate into main branch on next release.

opened issue for more permanent fix:
https://github.com/wehr-lab/autopilot/issues/33

thanks again for flagging this :)

closed 2020-10-17 by sneakers-the-rat

sneakers-the-rat, 2020-10-25:
merged in v0.3.3
pypi: https://pypi.org/project/auto-pi-lot/0.3.3/
release: https://github.com/wehr-lab/autopilot/releases/tag/v0.3.3
changelog: https://docs.auto-pi-lot.com/en/latest/changelog/v0.3.0.html#v0-3-3-october-25-2020