auto-pi-lot/autopilot #24: Document pilot connection & setup

sneakers-the-rat, 2020-10-14:
Received in discussion board: https://groups.google.com/g/autopilot-users/c/JvWIPpYY0TI/m/MtKAj0h-BwAJ

Hi Jonny,

First up, this seems like an amazing framework and I can't wait to get started! And alert: I'm an experimentalist, so a lot of things are new to me figuring out this - which might just be a good thing for providing n00b feedback on the docs. ;-)

I now have Autopilot installed on both a Terminal (Mac) and a Pilot. However, my knowledge is limited when it comes to connecting the two, and right now there's not much help to get from the docs. I have a vague idea that there's some ssh involved (about which I know pretty much 0), or is it simpler than that? Do they need to be connected to the same WiFi network (mine cannot currently where I am stationed)?

An aside: A later request for updating the docs could also be a brief mention of how to install and use virtualenv, in case people are not familiar with this way of working. This was the case for me so I had to figure that out elsewhere and hope for the best (works now, but might be very helpful for other first-timers. Also, it sort of works outside a virtual environment, but the Pilot doesn't as the welcome_msg.txt apparently is missing).

Thanks again, and hope you can help me out here!
Cheers,
[name removed in case they don't want it here!!!]

cxrodgers, 2020-12-13:
I recently went through this initial set up process and learned some things that I think would help other new users (especially non-experts) interpret the installation documentation.

- include the necessary commands to create the virtualenv
- indicate that MSGPORT on the terminal must equal PUSHPORT on the pilot
- some firewall hints
- gentler introduction to how the menu system for setup_autopilot works
- what needs to be installed on pilot and what needs to be installed on terminal
- how to start terminal and pilot for the first time and test that they can see each other

It looks like this info is mostly or all contained in 
https://github.com/wehr-lab/autopilot/blob/master/docs/guide.installation.rst

@sneakers-the-rat Can I take a stab at editing guide.installation.rst and send a PR for your review?

sneakers-the-rat, 2020-12-13:
These are definitely good additions. feel free to PR -- i'll be rolling up a few updates from `dev` soon and so we can include those in the next release :) If you don't have time to do it i'll definitely add these then.

sneakers-the-rat, 2020-12-13:
I think we might want to make a separate docs page for configuration, as the installation page will eventually get a bit overloaded with it, and after i've overhauled `prefs` recently there's a lot to add (but actually a sensible way to document it now!!)

cxrodgers, 2020-12-14:
Great, PR created!  https://github.com/wehr-lab/autopilot/pull/51
If you have any comments or think it should be done differently, feel free to edit the PR directly or just let me know what to change.

It occurs to me that it might be worthwhile to change the setup script for the Terminal to flip the MSGPORT/PUSHPORT defaults. It doesn't seem like there'd ever be a reason for the Terminal and Pilot agents to use the same defaults here (unless I'm missing something).

cxrodgers, 2020-12-14:
agreed that a separate Configuration page may be necessary.. As I'm getting my hands dirty, I'm finding it easier to start with your template prefs.json and get "free water" working, rather than figuring out the "add hardware" wizard just yet. We could consider adding this as a tutorial with the goal to get "free water" working and learn some things about autopilot along the way.

sneakers-the-rat, 2020-12-14:
yes, in previous instantiations of the setup routine defaults were split out by different agent types, but that proved to be cumbersome and I didn't re-implement when i re-designed prefs (though it is possible, i did include a `SCOPE` parameter to default prefs that could be used for that). I think the real solution is ultimately to redesign the way the networking modules work, the current system is pretty fragile, requires a lot of manual hardcoding to connect objects, and holds back some general things I want to be possible with the system (eg. being able to connect objects by using a simple string like `gpio_obj.state.connect('agent_id.object_group.object_id')` without needing to specifically know IP addresses, etc). Ideally there would be one expected port that's shared between agents to do an initial connection, all objects would have some networking object that automatically maps their `id` to a unique port, and the swarm keeps an updated table of objects available to it by announcing them when they're created. 

created a stub of the issue here: https://github.com/wehr-lab/autopilot/issues/48

the hardware wizard is a little clumsy, you're right, the cli gui might not be the right call there -- it tool is pretty brittle and i don't think the library is even maintained anymore -- but i wanted some intermediate for people controlling pis over ssh that wasn't just manually editing `prefs.json` (though that's valid and often what I do as well). The gui code for the library holds back a lot and I think this is one of those times, I think what might be good is to have a way to edit prefs on connected pilots from the terminal, but need to basically rebuild the entire gui module because it was one of the first things I programmed when i really didn't know what i was doing :p

i'll check out the pr later tonight or tomorrow, thanks for contributing much needed clarity :)

sneakers-the-rat, 2020-12-14:
merged the PR, will follow up with some additions i flagged for myself. closing this. thanks again :)

closed 2020-12-14 by sneakers-the-rat

mikewehr, 2020-12-14:
Chris, thanks for your helpful contributions to the docs and for your general comments!
-Mike