auto-pi-lot/autopilot #32: Minor Docs Tracker

sneakers-the-rat, 2020-10-17:
Track minor docs improvements not worth starting their own issue over

[ ] use of `~/.autopilot` file to bootstrap location of autopilot user dir

mikewehr, 2020-12-02:
Super confusing whether  "Board-numbered GPIO pin" refers to GPIO number (GPIO0-27) or pin number (1-40), there's got to be a way to spell this out so it's crystal clear in the docs

sneakers-the-rat, 2020-12-08:
"board-numbering" is the canonical way of referring to, well, the board number (the position of the pin on the board).

this is currently documented here:
https://docs.auto-pi-lot.com/en/dev/autopilot.hardware.html
[image]

and in the `pin_bcm` attribute here: https://docs.auto-pi-lot.com/en/dev/autopilot.hardware.gpio.html#autopilot.hardware.gpio.GPIO

I've added a link to a stackexchange post on the matter every time the phrase is mentioned in docstrings, and have made an additional note at the top of the gpio page here: https://github.com/wehr-lab/autopilot/commit/ebc39ba6efb3f0a47d29bcfdbf780c3feeb902a6

which should be reflected here: https://docs.auto-pi-lot.com/en/dev/autopilot.hardware.gpio.html

mikewehr, 2021-01-21:
basic debugging tutorial with tips & tricks and example workflows

sneakers-the-rat, 2022-02-09:
Document behavior of triggers re: advancing stage when not desired.

https://github.com/wehr-lab/autopilot/issues/148#issuecomment-1033255537

sneakers-the-rat, 2022-02-09:
all the various config files and the structure of the user directory should get its own page since it's a central concept that a lot of people (justifiably) miss, eg. what is a `pilot_db.json` is that documented anywhere? 

re: https://github.com/wehr-lab/autopilot/issues/148#issuecomment-1033255537

sneakers-the-rat, 2022-02-09:
add security notice -- this software is intended to be used on a local area network not exposed to the public internet and has not been hardened in any way, use at your own risk!

mikewehr, 2022-02-12:
installation instructions (here: https://docs.auto-pi-lot.com/en/latest/guide/installation.html)
do not clarify the best practices for having a separate git folder with the autopilot code, and a user autopilot folder with prefs, protocols, data, etc.
Jonny has commented with surprise more than once at my peculiar installation hierarchy in which there is an autopilot code directory nested inside my autopilot user directory (rather than in a segregated git directory). 
I have now installed autopilot from scratch a half dozen times, following the instructions on the installation page to the letter, and I end up with my peculiar nested file hierarchy every time. I agree that this weird nested structure is problematic even though it actually works ok.
I think the instructions need to include a couple extra steps recommending an appropriate sequence of mkdir ~/autopilot, mkdir ~/git, cd ~/git, git clone, pip3 install -e ., etc

mikewehr, 2022-02-12:
terminal needs an icon

sneakers-the-rat, 2022-02-12:
> installation instructions (here: https://docs.auto-pi-lot.com/en/latest/guide/installation.html) do not clarify the best practices for having a separate git folder with the autopilot code, and a [...]

definitely need a whole page on user directory structure a la

> all the various config files and the structure of the user directory should get its own page since it's a central concept that a lot of people (justifiably) miss, eg. what is a `pilot_db.json` is that documented anywhere?
> 
> re: [#148 (comment)](https://github.com/wehr-lab/autopilot/issues/148#issuecomment-1033255537)

sneakers-the-rat, 2022-02-21:
use and configuration of the systemd daemon

sneakers-the-rat, 2022-03-03:
use this to document the structure of the wiki ... on the wiki

https://github.com/wehr-lab/autopilot/issues/127#issuecomment-906794608

sneakers-the-rat, 2022-07-30:
Add to wiki https://github.com/auto-pi-lot/autopilot/discussions/182#discussion-4264149