auto-pi-lot/autopilot #33: Programmatic Setup

sneakers-the-rat, 2020-10-17:
Be able to specify setup options from the cli as flags, or from some portable environment description file that includes prefs as well as any scripts to be run.

- Fully resolves https://github.com/wehr-lab/autopilot/issues/23 by allowing setup to be run without terminal window being a minimum size
- makes multiple setups much easier
- finalizes portability of experiments by allowing full environment recreation without need for manual intervention.

roaldarbol, 2020-11-02:
If I try to use a different user directory than the default during setup, I get the following error - I think there might be a hardcoding of '~/autopilot/' associated with the creation of `pilot_db.json`:

`(venv) (base) roaldarbol@roaldarbol autopilot_dev % python3 -m autopilot.setup.setup_autopilot`
`Traceback (most recent call last):`
`File "/usr/local/anaconda3/lib/python3.7/runpy.py", line 193, in _run_module_as_main`
`   "__main__", mod_spec)`
` File "/usr/local/anaconda3/lib/python3.7/runpy.py", line 85, in _run_code`
`    exec(code, run_globals)`
`  File "/Users/roaldarbol/Documents/nivenlab/autopilot_dev/autopilot/autopilot/setup/setup_autopilot.py", line 571, in` `<module>`
`    with open(terminal['PILOT_DB'], 'w') as pilot_db_file:`
`FileNotFoundError: [Errno 2] No such file or directory: '/Users/roaldarbol/Documents/nivenlab/autopilot/pilot_db.json'`

sneakers-the-rat, 2020-11-02:
Opened this as its own issue: 
https://github.com/wehr-lab/autopilot/issues/40

If you set the `BASEDIR` field in the setup routine it should work, though it may also be a directory creation error. opened a separate issue to describe and fix:

https://github.com/wehr-lab/autopilot/issues/40

sneakers-the-rat, 2021-07-30:
This is done here: https://github.com/wehr-lab/autopilot/commit/ccec973f68e83ed4320fb7addc295a9117583bc5

closed 2021-07-30 by sneakers-the-rat