auto-pi-lot/autopilot #34: curses bug in setup script

evilrobotxoxo, 2020-10-25:
```
python3 -m autopilot.setup.setup_autopilot
```
reproducibly crashes if you run it in an SSH terminal window that is too narrow. This is admittedly a minor issue, but it was an annoyance that took me a while to figure out.

covid-confessions, 2020-10-25:
ah yes, fixed in https://github.com/wehr-lab/autopilot/issues/23 but i haven't pulled it back to main & re-released yet. was waiting on testing a few other changes that have been made in that branch but that's done at this point so i'll pull back now. sorry for making you re-solve this one!!

sneakers-the-rat, 2020-10-25:
Ah yes, fixed with informative error message in https://github.com/wehr-lab/autopilot/issues/23 and related to a final resolution in https://github.com/wehr-lab/autopilot/issues/33 . Was waiting to test some other changes that had been made in that branch, but that's been done and so I'll merge & re-release shortly.

closed 2020-10-25 by sneakers-the-rat

sneakers-the-rat, 2020-10-25:
merged in v0.3.3
pypi: https://pypi.org/project/auto-pi-lot/0.3.3/
release: https://github.com/wehr-lab/autopilot/releases/tag/v0.3.3
changelog: https://docs.auto-pi-lot.com/en/latest/changelog/v0.3.0.html#v0-3-3-october-25-2020