auto-pi-lot/autopilot #36: ports not responding

lucasogit, 2020-10-28:
After running boxes for around 30 minutes plus the ports start acting weird and ultimately stop responding. Example: started up test protocol -> let run for 45 min -> first couple of port inputs worked -> lights got wonky and 3rd poke and beyond were not responsive. 
Terminal:
[portsnotworking.txt](https://github.com/wehr-lab/autopilot/files/5449286/portsnotworking.txt)

sneakers-the-rat, 2020-10-28:
I have a feeling i know what this might be... either script creation or storing too many events. will need pilot logs to debug, but will come into lab tmrw to check on.

lucasogit, 2020-10-29:
[port.not.working.terminal.txt](https://github.com/wehr-lab/autopilot/files/5455881/port.not.working.terminal.txt)
[pilot_logs.tar.gz](https://github.com/wehr-lab/autopilot/files/5455882/pilot_logs.tar.gz)
Bug happened again after stopping protocol after running for an hour and then starting the next mouse's protocol.

sneakers-the-rat, 2020-10-29:
aha this is related to getting timestamps in pigpio:

```
Oct 28 18:40:00 pilot2 launch_autopilot.sh[726]: Exception in thread Thread-709:
Oct 28 18:40:00 pilot2 launch_autopilot.sh[726]: Traceback (most recent call last):
Oct 28 18:40:00 pilot2 launch_autopilot.sh[726]:   File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
Oct 28 18:40:00 pilot2 launch_autopilot.sh[726]:     self.run()
Oct 28 18:40:00 pilot2 launch_autopilot.sh[726]:   File "/home/pi/git/autopilot/venv/lib/python3.7/site-packages/pigpio.py", line 1230, in run
Oct 28 18:40:00 pilot2 launch_autopilot.sh[726]:     tick = self.ticks_to_timestamp(tick)
Oct 28 18:40:00 pilot2 launch_autopilot.sh[726]:   File "/home/pi/git/autopilot/venv/lib/python3.7/site-packages/pigpio.py", line 1208, in ticks_to_timestamp
Oct 28 18:40:00 pilot2 launch_autopilot.sh[726]:     timestamp = datetime.fromtimestamp(timestamp).isoformat()
Oct 28 18:40:00 pilot2 launch_autopilot.sh[726]: OverflowError: timestamp out of range for platform time_t
Oct 28 18:40:00 pilot2 launch_autopilot.sh[726]: Exception in thread Thread-710:
Oct 28 18:40:00 pilot2 launch_autopilot.sh[726]: Traceback (most recent call last):
Oct 28 18:40:00 pilot2 launch_autopilot.sh[726]:   File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
Oct 28 18:40:00 pilot2 launch_autopilot.sh[726]:     self.run()
Oct 28 18:40:00 pilot2 launch_autopilot.sh[726]:   File "/home/pi/git/autopilot/venv/lib/python3.7/site-packages/pigpio.py", line 1230, in run
Oct 28 18:40:00 pilot2 launch_autopilot.sh[726]:     tick = self.ticks_to_timestamp(tick)
Oct 28 18:40:00 pilot2 launch_autopilot.sh[726]:   File "/home/pi/git/autopilot/venv/lib/python3.7/site-packages/pigpio.py", line 1208, in ticks_to_timestamp
Oct 28 18:40:00 pilot2 launch_autopilot.sh[726]:     timestamp = datetime.fromtimestamp(timestamp).isoformat()
Oct 28 18:40:00 pilot2 launch_autopilot.sh[726]: OverflowError: timestamp out of range for platform time_t
Oct 28 18:40:00 pilot2 launch_autopilot.sh[726]: INFO:hardware.POKES_R:POKES_R Logging Initiated
Oct 28 18:40:00 pilot2 launch_autopilot.sh[726]: Exception in thread Thread-711:
Oct 28 18:40:00 pilot2 launch_autopilot.sh[726]: Traceback (most recent call last):
Oct 28 18:40:00 pilot2 launch_autopilot.sh[726]:   File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
Oct 28 18:40:00 pilot2 launch_autopilot.sh[726]:     self.run()
Oct 28 18:40:00 pilot2 launch_autopilot.sh[726]:   File "/home/pi/git/autopilot/venv/lib/python3.7/site-packages/pigpio.py", line 1230, in run
Oct 28 18:40:00 pilot2 launch_autopilot.sh[726]:     tick = self.ticks_to_timestamp(tick)
Oct 28 18:40:00 pilot2 launch_autopilot.sh[726]:   File "/home/pi/git/autopilot/venv/lib/python3.7/site-packages/pigpio.py", line 1208, in ticks_to_timestamp
Oct 28 18:40:00 pilot2 launch_autopilot.sh[726]:     timestamp = datetime.fromtimestamp(timestamp).isoformat()
Oct 28 18:40:00 pilot2 launch_autopilot.sh[726]: OverflowError: timestamp out of range for platform time_t

```

which i have sorta hacked into my own fork of pigpio here: https://github.com/sneakers-the-rat/pigpio/blob/c481e0f309d7d182aa697c43800d305d819a24b7/pigpio.py#L5310

in order to be able to get absolute timestamps in isoformat rather than system ticks... Clearly something is wrong with the way i'm getting the timestamps, but in the intermediate time I can just add a catch to fallback to ticks with a warning. dont have time to chase this down tonight but can do when i'm in the lab on friday afternoon

sneakers-the-rat, 2020-11-04:
temporarily fixed here -- https://github.com/sneakers-the-rat/pigpio/commit/6625928f31c5bf0179663e7159a4766a25db6586

the issue seems to be that when the tick clock rolls over the new offset isn't communicated to the callback threads. should be simple to implement, raised an issue to keep track of it here: https://github.com/sneakers-the-rat/pigpio/issues/2

behavior now is to fallback to returning ticks and raise a warning with info that should be diagnostic of the problem.

closed 2020-11-04 by sneakers-the-rat