auto-pi-lot/autopilot #37: boxes not being recognized when auotpilot opens

lucasogit, 2020-10-28:
Ocasionally I'll open autopilot and a bunch of the boxes will be offline. Usually takes resetting boxes, network and computer a couple of times to get all the boxes online at the same time. 
Terminal:
[offlineboxes.txt](https://github.com/wehr-lab/autopilot/files/5449290/offlineboxes.txt)

sneakers-the-rat, 2020-10-28:
need to reinstate pinging and heartbeating!!! i saw this a few times with boxes that were slow to start networking. this should be fixed with periodic pings when no message has been received in x time

sneakers-the-rat, 2021-07-22:
This was fixed enough until we overhaul networking that i'm satisfied with it and has been a QoL improvement:

https://github.com/wehr-lab/autopilot/commit/dc9833ae827301219d50620fa8e1b22266bb222d

https://github.com/wehr-lab/autopilot/commit/733131a787cc00b459748aa2248c2ae3bb82b91a

https://github.com/wehr-lab/autopilot/commit/3dd5206e6f9b874e6aea234398a47f875e13024b

closed 2021-07-22 by sneakers-the-rat