auto-pi-lot/autopilot #41: sound failing/boxes stop responding

lucasogit, 2020-11-12:
There are 2 issues contained in the attached log files.
Box 4: sound cut out completely at some point -> restarted box in autopilot into the ports not working/box not responding
Box 7: ports/box stopped responding at some point during running
Box 0: ports/box stopped responding at some point during running -> restarted box in autopilot and still not responding

Log files:
[pilot_logs.tar.gz](https://github.com/wehr-lab/autopilot/files/5527935/pilot_logs.tar.gz)
[box_error_11.11.20.txt](https://github.com/wehr-lab/autopilot/files/5527940/box_error_11.11.20.txt)

sneakers-the-rat, 2020-11-14:
sound failure was a jackd overlap issue -- this one missed having its installation in /usr/local/bin removed, so it was having symbol mismatches between /usr/lib and /usr/local/lib versions. removed differing versions and reinstalled and fixed.

boxes 0 and 7 should be simple, on instantiation of a task, we get this:

```
Nov 11 15:54:56 pilot7 launch_autopilot.sh[739]: ERROR:task.<class 'autopilot.tasks.nafc.Nafc_Gap'>:Pin could not be instantiated - Type: PORTS, Pin: R
Nov 11 15:54:56 pilot7 launch_autopilot.sh[739]: Traceback (most recent call last):
Nov 11 15:54:56 pilot7 launch_autopilot.sh[739]:   File "/home/pi/git/autopilot/autopilot/tasks/task.py", line 151, in init_hardware
Nov 11 15:54:56 pilot7 launch_autopilot.sh[739]:     hw = handler(hw_args, name=hw_name)
Nov 11 15:54:56 pilot7 launch_autopilot.sh[739]:   File "/home/pi/git/autopilot/autopilot/hardware/gpio.py", line 1238, in __init__
Nov 11 15:54:56 pilot7 launch_autopilot.sh[739]:     self.duration = duration
Nov 11 15:54:56 pilot7 launch_autopilot.sh[739]:   File "/home/pi/git/autopilot/autopilot/hardware/gpio.py", line 1253, in duration
Nov 11 15:54:56 pilot7 launch_autopilot.sh[739]:     self.store_series(id="open", values=self.on, durations=self._duration, unit="ms", finish_off=True, repeat=None)
Nov 11 15:54:56 pilot7 launch_autopilot.sh[739]:   File "/home/pi/git/autopilot/autopilot/hardware/gpio.py", line 472, in store_series
Nov 11 15:54:56 pilot7 launch_autopilot.sh[739]:     script_id = self.pig.store_script(series_script)
Nov 11 15:54:56 pilot7 launch_autopilot.sh[739]:   File "/home/pi/git/autopilot/venv/lib/python3.7/site-packages/pigpio.py", line 4372, in store_script
Nov 11 15:54:56 pilot7 launch_autopilot.sh[739]:     self.sl, _PI_CMD_PROC, 0, 0, len(script), [script]))
Nov 11 15:54:56 pilot7 launch_autopilot.sh[739]:   File "/home/pi/git/autopilot/venv/lib/python3.7/site-packages/pigpio.py", line 1031, in _u2i
Nov 11 15:54:56 pilot7 launch_autopilot.sh[739]:     raise error(error_text(v))
Nov 11 15:54:56 pilot7 launch_autopilot.sh[739]: pigpio.error: 'no more room for scripts'
```

so should be a straightforward patch -- currently GPIO objects don't clear their scripts when they are removed. they should do that. otherwise they stay on in pigpiod which reserves a very small amount of memory for them.

changing bug name accordingly.

sneakers-the-rat, 2020-11-21:
* Increased max number of scripts to 256 here: https://github.com/sneakers-the-rat/pigpio/commit/ad974f99925ae709ef3fd175f54899888b0c6785
* stop and delete all scripts on gpio object release: https://github.com/wehr-lab/autopilot/commit/e8218304bd7ef2e13d2adfc236f3e781abea5f78

closed 2020-11-21 by sneakers-the-rat