auto-pi-lot/autopilot #46: trying to launch terminal

mikewehr, 2020-11-30:
$ source launch_autopilot.sh
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/Users/mikewehr/Documents/Analysis/autopilot/autopilot/core/terminal.py", line 885, in <module>
    ex = Terminal()
  File "/Users/mikewehr/Documents/Analysis/autopilot/autopilot/core/terminal.py", line 159, in __init__
    prefs.add('INVOKER', self.invoker)
  File "/Users/mikewehr/Documents/Analysis/autopilot/autopilot/prefs.py", line 414, in add
    _PREFS[param] = value
  File "<string>", line 2, in __setitem__
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/multiprocessing/managers.py", line 818, in _callmethod
    conn.send((self._id, methodname, args, kwds))
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/multiprocessing/connection.py", line 206, in send
    self._send_bytes(_ForkingPickler.dumps(obj))
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/multiprocessing/reduction.py", line 51, in dumps
    cls(buf, protocol).dump(obj)
TypeError: can't pickle Invoker objects

mikewehr, 2020-11-30:
also how am I supposed to launch autopilot? This is macos. 

Transporter:autopilot mikewehr$ autopilot
-bash: autopilot: command not found
Transporter:autopilot mikewehr$ launch_autopilot.sh
-bash: launch_autopilot.sh: command not found
Transporter:autopilot mikewehr$ ls -l launch_autopilot.sh
-rwxrwxr-x  1 mikewehr  staff  87 Nov 24 18:22 launch_autopilot.sh
Transporter:autopilot mikewehr$

sneakers-the-rat, 2020-11-30:
it appears as if you've pulled from the dev branch and i'm not quite finished with changes to prefs... quick on the draw, i just wrote that this afternoon ;) how about this, i'll make a 'deploy' branch that's the one for code deployed to our pi array, and then i'll keep working on 'dev' for now so we stop having instability in the behavior system (used to be the case that i was the only one pulling code, but makes sense to change protocol now)

keeping issue open to remind me to add a script option at installation to add `f"alias autopilot={launch_script}"` to bash profile -- the more permanent fix will be to use setuptools entrance hooks, but need to massage the external process management a bit before abandoning the (admittedly janky) bash script entrypoint for now.

sneakers-the-rat, 2020-11-30:
Tell you what - the patch has now trisected... this is a problem best fixed by implementing registries, which relies on fixing prefs, because the task is programmed in the current style but is a sorta textbook example of a 'plugin' task (and the work is all mushed in that branch).  working on it now, give me thru tmrw to clean up and all will become clear

sneakers-the-rat, 2020-12-02:
implemented: https://github.com/wehr-lab/autopilot/commit/211919b05922e18a85d8ef6216973f4000fd32c5

closed 2020-12-02 by sneakers-the-rat