auto-pi-lot/autopilot #49: autopilot GUI not responding

lucasogit, 2020-12-11:
While stopping the file of a running mouse, autopilot gui went gray and stopped responding in any way. Tasks were still going.

Logs:
[pilot_logs.tar.gz](https://github.com/wehr-lab/autopilot/files/5676330/pilot_logs.tar.gz)
[autopilot_not_responding_terminal.txt](https://github.com/wehr-lab/autopilot/files/5676331/autopilot_not_responding_terminal.txt)

sneakers-the-rat, 2020-12-11:
was there some other dialog window open, eg. the weight entry window?

sneakers-the-rat, 2020-12-13:
I'm assuming that's what happened, that the weight dialogue got popped under the main window, and since the dialogue is modal it blocks the terminal. 

changing the name of his issue accordingly

sneakers-the-rat, 2021-07-22:
this actually was a bug in handling keys as a list instead of a generator when switching from 2 to 3. in any case being modal is sort of important for how the gui windows work now, and the fix here will come after we a) fix networking and b) split up the GUI so it doesn't share state everwhere and instead uses message passing as god intended it to.

closed 2021-07-22 by sneakers-the-rat