auto-pi-lot/autopilot #56: Let Terminal GUI remember its previous size

cxrodgers, 2020-12-30:
Current behavior is that the Terminal GUI covers the entire desktop, which probably makes sense for certain use-cases but can be annoying in others. I see there are some comments about infinitely expanding buttons so I imagine that was the rationale for this choice.

Perhaps a more user-friendly behavior would be to remember the previous size and restore it? Can autopilot/prefs.json be a good place to store information like last window size or is there a better place? I can work on this if people think it makes sense.

On a related note I fixed some Qt DeprecationWarnings on my end with this commit:
https://github.com/cxrodgers/autopilot/commit/f22002ec929644930b49114e8392a2449d4367d7
I can include that in a PR too.

sneakers-the-rat, 2020-12-30:
Very generous of you to grant my inability to figure out an infinitely expanding button bug as a choice :p  

Yes this sounds like a great idea. 

Unhelpfully, the Qt foundation seems to have moved to PySide6 and dropped hosting docs for PySide2, i'm building them now but assuming PySide2 has similar methods...

The `QWidget` class should have `saveGeometry` and `restoreGeometry` methods (https://doc.qt.io/qtforpython/PySide6/QtWidgets/QWidget.html#PySide6.QtWidgets.PySide6.QtWidgets.QWidget.saveGeometry and https://doc.qt.io/qtforpython/PySide6/QtWidgets/QWidget.html#PySide6.QtWidgets.PySide6.QtWidgets.QWidget.restoreGeometry ) and there is an example in the docs about how to store state in the docs.

What I think would work is having a `pref` with `Scope.TERMINAL` that lets people choose between fullscreen and remember geometry, if set to remember geometry then save geometry to a file in the user directory on exit, load when opening, etc.

The GUI code is easily the worst code in the library at this point and holds a lot of things back. wrote a lot of it when i was first learning python :\. It needs a major refresh, but yes for now would welcome this improvement.

cxrodgers, 2020-12-31:
cool will give it a shot

mikewehr, 2021-01-17:
Did anybody ever try to implement this (remember terminal size)?

cxrodgers, 2021-01-17:
Yes, I've implemented this, and can issue a PR. 
https://github.com/wehr-lab/autopilot/compare/dev...cxrodgers:dev__terminal_winsize
I should get feedback from you all first though, in case you have suggestions. Here was my approach:

- As Jonny suggested, I defined a new pref with Scope.TERMINAL called TERMINAL_WINSIZE_BEHAVIOR that takes one of three values: "maximum", "remember", or "moderate". 
- For "moderate", the terminal always opens to a 1000x400 pixel size. For "maximum", the terminal always fills the primary display. For "remember", the terminal opens to whatever size it was last time it was closed. The memory is implemented using QSettings restoreGeometry function that jonny linked to. If there is no stored setting, it defaults to "moderate".
- The default behavior is "remember". I recommend this default, because I think the typical user's first experience will be with just one or two Pis, so they don't need maximum screen. And from then on they can resize it, and the Terminal will just remember whatever the last size was. For people like you all with (I'm imagining) large flocks of Pis and therefore want maximum, you would just have to add the pref to be "maximum" in prefs.json, or maximize it once and it would remember from then on.

What do you think?

sneakers-the-rat, 2021-02-10:
closed here : https://github.com/wehr-lab/autopilot/pull/70

sneakers-the-rat, 2021-02-10:
Thanks again!!!!

closed 2021-02-10 by sneakers-the-rat