auto-pi-lot/autopilot #84: no sound on box 3

lucasogit, 2021-03-09:
When I put 37 into box 3 there was sound. When I took him out, the sound had stopped playing. This was on the laser step of Mike's gpr26 laser task.

Logs:
[pilot_logs.tar.gz](https://github.com/wehr-lab/autopilot/files/6105982/pilot_logs.tar.gz)
[no_sound_box3.txt](https://github.com/wehr-lab/autopilot/files/6105983/no_sound_box3.txt)

sneakers-the-rat, 2021-03-18:
It looks like the jack client is closing prematurely or having some other error connecting to jackd, 
```
Mar 08 18:47:49 pilot3 launch_autopilot.sh[736]: JackEngine::XRun: client = jack_client-01 was not finished, state = Running
Mar 08 18:47:49 pilot3 launch_autopilot.sh[736]: JackAudioDriver::ProcessGraphAsyncMaster: Process error
Mar 08 18:47:49 pilot3 launch_autopilot.sh[736]: JackAudioDriver::ProcessGraphAsyncMaster: Process error
Mar 08 18:47:49 pilot3 launch_autopilot.sh[736]: JackAudioDriver::ProcessGraphAsyncMaster: Process error
```

but I can't tell what's happening because it looks like `LOGLEVEL` is set to warning in the pilot. Can you change `LOGLEVEL` to `'DEBUG'` and reproduce problem & send logs?

lucasogit, 2021-03-19:
Updated logs with debug mode on:
[no_sound_box2.txt](https://github.com/wehr-lab/autopilot/files/6168085/no_sound_box2.txt)
[pilot_logs.tar.gz](https://github.com/wehr-lab/autopilot/files/6168086/pilot_logs.tar.gz)

Note: Issue occurred on box 2 this time around.

sneakers-the-rat, 2021-03-29:
could be as simple as an uninitialized attribute:

```
Mar 18 15:52:24 pilot2 launch_autopilot.sh[734]: Exception in thread Thread-27:
Mar 18 15:52:24 pilot2 launch_autopilot.sh[734]: Traceback (most recent call last):
Mar 18 15:52:24 pilot2 launch_autopilot.sh[734]:   File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
Mar 18 15:52:24 pilot2 launch_autopilot.sh[734]:     self.run()
Mar 18 15:52:24 pilot2 launch_autopilot.sh[734]:   File "/usr/lib/python3.7/threading.py", line 865, in run
Mar 18 15:52:24 pilot2 launch_autopilot.sh[734]:     self._target(*self._args, **self._kwargs)
Mar 18 15:52:24 pilot2 launch_autopilot.sh[734]:   File "/home/pi/git/autopilot/autopilot/stim/sound/sounds.py", line 247, in wait_trigger
Mar 18 15:52:24 pilot2 launch_autopilot.sh[734]:     self.trigger()
Mar 18 15:52:24 pilot2 launch_autopilot.sh[734]:   File "/home/pi/git/autopilot/autopilot/tasks/nafc.py", line 753, in stim_end
Mar 18 15:52:24 pilot2 launch_autopilot.sh[734]:     if self.laser_script is not None:
Mar 18 15:52:24 pilot2 launch_autopilot.sh[734]: AttributeError: 'Nafc_Gap_Laser' object has no attribute 'laser_script'
```

pushed an update here: https://github.com/wehr-lab/autopilot/commit/2d547136f4b45b9b96d4f574f02802fb5f33fc64

update the pis and reopen if problem persists :)

closed 2021-03-29 by sneakers-the-rat

lucasogit, 2021-03-31:
Same issue. No sound at the end of running for box 2.

Logs: 
[box2_no_sound.txt](https://github.com/wehr-lab/autopilot/files/6239776/box2_no_sound.txt)
(pilot_logs.tar.gz was too big to post in a comment. I put it slack as a reply to the bot autoresponse to this comment)

sneakers-the-rat, 2021-04-05:
yeah nothing that i can see that would have caused the problem. i can't really tell when the error happened, i'm just looking at the last day of logs, but would want to know when you noticed the sound was off. I'll be in tomorrow, let's see if we can recreate the problem then.