auto-pi-lot/autopilot #94: Support for stereo (two-channel) audio output

cxrodgers, 2021-05-27:
My understanding from reading the code is that stereo (two-channel) audio output is not really supported in autopilot right now. I have been working on implementing this in my own fork. It isn't quite ready for a PR yet, but I wanted to share my approach and collect feedback from anyone who is interested in this functionality.

__Here is how (I think) things currently stand in autopilot:__
There are prefs OUTCHANNELS and NCHANNELS that relate to multi-channel output. I believe that they ultimately serve just to hook up the same "mono" audio stream to one or both physical outputs on the hifiberry. The sound generating classes seem to generate only mono sounds.

__Here are the scenarios and use-cases that I would like to support:__
* One speaker hooked up to one channel
* Two speakers hooked up to two channels, but always playing the same thing ("mono")
* Two speakers hooked up to two channels, playing different things ("stereo")
* Two speakers hooked up to two channels, playing different things, but reversed ("reversed")

The last scenario would be used in the case where you've hooked everything up but realized you swapped the speakers by accident, and just want to flip a bit in software instead of physically rewiring things.

I can't really imagine a scenario where we have more than two channels of audio output. I am actually implementing now 8-channel output, and the way I'm doing it is with 4 Pis and 4 Hifiberrys. I would suggest that we don't attempt to code in support for >2 channel output in software until someone comes up with a hardware solution or proposal, since I don't even know what that would like right now.

__My proposal__
First, I'm proposing to add a `channel` argument to the sound-generating classes. It would have the following behavior:
* `channel=None`: mono (1-dimensional) waveform, the same as it is now
* `channel=0`: stereo output, with channel 0 containing a waveform and channel 1 containing zeros
* `channel=1`: stereo output, with channel 1 containing a waveform and channel 0 containing zeros

Second, I think we could simplify the logic by implementing each use-case with OUTCHANNELS (and dropping NCHANNELS). I propose the following meaning for OUTCHANNELS:
* [0] or [1]: single output to the corresponding speaker
* [0, 1]: stereo output. Whether they play mono or stereo sounds would be determined by the sound generating classes (the `channel` argument above). 
* [1, 0]: reversed stereo output. No change to the sound generating classes would be needed here, just flip this bit in prefs to swap the speakers.

The default behavior would be `channel=None` in the sound generating classes (which generates mono, 1d output just as they current do) and [0, 1] for OUTCHANNELS. That would send mono output to all speakers, or if someone generates stereo sound, that would work right away without changing OUTCHANNELS.

__Status__
Currently implemented in https://github.com/cxrodgers/autopilot/tree/stereo_sound

A comparison vs the "dev" branch:
https://github.com/wehr-lab/autopilot/compare/dev...cxrodgers:stereo_sound

I believe that it's working on my end. I'm a little unsure about the logic in jackclient.py because there are a lot of conditionals relating to continuous sound (or not), currently playing sound (or not), and the queues. I'll check more carefully.

A few stray observations:
* I think there is some Matlab-style indexing lurking in the default pref for OUTCHANNELS of [1, 2], this should almost surely be [0, 1].
* I'm having trouble testing the sound-generating classes because I don't know how to instantiate a jackclient outside of a task, and that jackclient is needed to get the sampling rate and so on. I have noticed a few weirdnesses with sounds, where if I generate the same sound twice it is two different lengths, one the requested length and one a multiple of the chunk size (1024). Still tracking this down

Please let me know if you have any suggestions! The PR should be ready soon :)

sneakers-the-rat, 2021-05-27:
Yes! Love this!!! let me see how i can help

Going through the whole lifecycle of the sound server:
- The `pilot` starts `jackd` and the `jack_server` here: https://github.com/wehr-lab/autopilot/blob/dc9833ae827301219d50620fa8e1b22266bb222d/autopilot/core/pilot.py#L587-L591  jackd is the daemon that needs to be running in the background, and `jack_server` is the client that interacts with it
- `start_jackd` function is just a very slim wrapper around a shell call to start `jackd`: https://github.com/wehr-lab/autopilot/blob/dc9833ae827301219d50620fa8e1b22266bb222d/autopilot/external/__init__.py#L98
- `jackd_server` is initialized in this method. The jack client has `outports` which are the 'client-facing' virtual ports that sound can be dumped into, in this case initializing only one `'out_0'` port is hardcoded. The `outports` are connected to physical ports, so eg. one `outport` can output to multiple physical output ports, etc. In the current version the single `outport` is connected to one (in the case that `OUTCHANNELS` is an int) or multiple (a list) channels. https://github.com/wehr-lab/autopilot/blob/dc9833ae827301219d50620fa8e1b22266bb222d/autopilot/stim/sound/jackclient.py#L168-L192
- the `jackd_server` registers a `process` callback that is periodically called to refill the buffer. That's this one, which as you note has a ton of conditional logic in it. The reason for that is that the interface between the sound objects and the jackd server is relatively 'dumb' -- they only have queue(s) and event flags. I think that method is pretty opaque and should be broken up, but it's a little tricky because that method has to be really fast because it's called so often and has to fill the buffer or you get xruns.  https://github.com/wehr-lab/autopilot/blob/dc9833ae827301219d50620fa8e1b22266bb222d/autopilot/stim/sound/jackclient.py#L220
- The critical thing in the `process` method that would need to be changed to support multichannel audio is to change the places where a simple transposed array is dumped into the single outport into patterns like this which iterate over the outports: https://github.com/wehr-lab/autopilot/blob/dc9833ae827301219d50620fa8e1b22266bb222d/autopilot/stim/sound/jackclient.py#L277 the problem i encountered with this before was some inexplicable audio garbling that i couldn't really explain, but always knew i would have to revisit.

I think you're right, `NCHANNELS` should go, and `OUTCHANNELS` should be reworked. I think what `OUTCHANNELS` ultimately should represent is the part of the client that maps the virtual `outports` to the physical output ports (confusing names sry). Once the server is booted, it only interacts with the virtual `outports` How about this:
- `None` is explicitly undefined, where the server doesn't map any ports. Seems weird, don't know exactly when one might want to boot a sound server but not map any ports, but seems like the most straightforward meaning for `None`
- a single `int` or a list of length 1 connects the 0th virtual `outport` to the physical output channel specified by the int.
- a list, say `[0, 3]` connects the 0th...nth `outport` to the channels specified, so in this case `client.outports[0].connect(0); client.outports[1].connect(3)`. 
- a dict, say `{0: [0, 1], 1: [2, 3]}` would connect virtual outports specified as keys to physical outports as ints or lists explicitly, so eg. one could map one audio stream to multiple speakers.

Unclear to me if it's better to have this kind of virtualization or if the virtual `outports` should try to be 1-1 with the physical ports (ie. to output to physical channel `1`, the `1`st outport should be used).

For the sound generation classes, I think you're right, they need a `channel` argument. I think it might be good to match the syntax of ^^, so stereo output being `channels=[0,1]`, and maybe we could give some simple aliases like `'mono' == 0` and `'stereo' == [0,1]`. Does it make more sense for the default to just play sound across all `outport`s? (sort of what i would expect from consumer speaker equipment, press play on a mono sound and it comes out both speakers) or to default to the 0th port?

The tricky thing i think would be how the `jackclient` receives and interprets multichannel sounds -- eg. if it receives a 1 dimensional array, but the array is supposed to go to `outchannel` 1 rather than 0, how would we indicate that? I think the current implementation of the client is pretty inflexible in that regard and should probably be reworked down the road to enable things that seem obviously desirable like eg. being able to start and stop sounds independently on different channels, playing multiple sounds at once, etc. but maybe out of scope for this issue.

I will take a look at your implementation asap, want to try to finish the networking fixes and get them pulled, as well as get 0.4.0 with the registry system out the door, but would absolutely welcome a PR on this :)

sneakers-the-rat, 2021-05-27:
there's an ambiguity there where one could play a stereo sound by mapping a single outport to two physical channels, or by making two outports that each map to one. I don't think that's necessarily a bad thing, but the former would probably be faster -- maybe just something to clarify in docs

cxrodgers, 2021-05-30:
I've successfully rediscovered your "inexplicable audio garbling" :) When I provide mono sound to my implementation, everything works. When I provide stereo sound, I get garbling and the xrun errors you mentioned:
```
JackEngine::XRun: client = jack_client-01 was not finished, state = Running
JackAudioDriver::ProcessGraphAsyncMaster: Process error
JackAudioDriver::ProcessGraphAsyncMaster: Process error
JackAudioDriver::ProcessGraphAsyncMaster: Process error
JackAudioDriver::ProcessGraphAsyncMaster: Process error
```
I thought maybe this was the increased time taken to write to two outports, but strangely enough, when I comment out all the code that writes to the second outport, the garbling persists. So I wonder if it is something about passing the double-size stereo data around that is slowing things down.

Two background-type questions:
* Some of your recent commits led me to believe that the proper way to install jackd2 is now via APT, not building from source, is that right? That's how I installed it.
* I am not doing `export JACK_NO_AUDIO_RESERVATION=1`. At some point I thought this was necessary but it turns out a reboot fixes that. Regardless, results are the same either way.

Higher latency is no problem for my application, so I tried increasing the -p and -n variables in the pref JACKDSTRING
`"JACKDSTRING": "jackd -P75 -p16 -t2000 -dalsa -dhw:sndrpihifiberry -p1024 -P -rfs -n3 -s &"`
But this actually seems to make things worse. 

I'll keep debugging :)

cxrodgers, 2021-05-30:
I found the problem!! The old padding code in Jack_Sound.chunk didn't work for 2-dimensional sound tables. If the last chunk needed to be padded by 400 samples, it would also be padded by 400 samples in the channel dimension (default behavior of `np.pad`), resulting it in an enormous 402-channel waveform, which slowed down jackclient. Somehow all this happened silently without actually raising any actual errors.

Anyway fixed here:
https://github.com/cxrodgers/autopilot/commit/2d3bd7309e6afe3469368a21f12d32fbb8ab2117

Stereo sound is ungarbled.

sneakers-the-rat, 2021-06-01:
omg OF COURSE. it's both impressive and hilarious that jack took the 402 channel sound like a champ and just went for it. thank you for chasing that down! looking forward to the PR, taking a look at that commit now I'm also obliged to thanks again for improving the comments & docstrings as you go.

re: jack installation -- there is this longstanding problem with jack provided through APT for the raspi. Long story short dbus doesn't really play nicely out of the box, and so reserving a soundcard for RT requires a lot of unnecessary work (related to your question re: `JACK_NO_AUDIO_RESERVATION`). For a brief period there was some claim on the raspi forums that this was fixed, and at the time the `scripts` functionality was really primitive, so I figured it would be easier to install from apt. This has turned out to *not* be the case and cause a ton of mysterious issues, and so i think the APT method should actually be removed, the `jackd_source` script (`python -m autopilot.setup.run_script jackd_source`) installs correctly and is what i recommend.

sneakers-the-rat, 2021-07-27:
closed with https://github.com/wehr-lab/autopilot/pull/102

closed 2021-07-27 by sneakers-the-rat