auto-pi-lot/autopilot #114: Change setuptools to poetry

sneakers-the-rat, 2021-07-30:
i am tired of working around the pretty bad dependency specification that relies on keeping a bunch of plaintext requirements file in sync with one another instead of using optional dependencies. Also a bunch of functionality that should be handled by a package manager is done in the distressingly long `setup_autopilot.py` script. I've been using poetry for all my other projects and love it. Lockfiles are great for reliability and very much in-line with having a particular environment specified for a particular git hash that we're trying to do here. This is also the direction that python packaging is going (away from `setup.py` and setuptools) and i see why.

One downside is that poetry packages don't allow editable `-e` installations right now. using a poetry shell is essentially equivalent, but it's still a foreign thing (and is still a little magical to me).

sneakers-the-rat, 2021-10-29:
Need to add ability to install optional sets of packages from setup menu, to avoid eg. https://github.com/wehr-lab/autopilot/issues/142

sneakers-the-rat, 2021-12-19:
Finished in: 
https://github.com/wehr-lab/autopilot/commit/871ff975d1be0b4231b3e121240588fb79829a83

and refined in following commits on the v0.5.0 branch

Also related to the first half of: https://github.com/wehr-lab/autopilot/issues/139

closed 2021-12-19 by sneakers-the-rat

cxrodgers, 2022-08-01:
Trying to update to 0.5.0 but unsure how to handle this change. I would like to do the equivalent of the old `pip install -e .`. This method is still mentioned in docs (https://docs.auto-pi-lot.com/en/latest/guide/installation.html) but I think it no longer works because there is no `setup.py`. 

Can you help me understand what the poetry equivalent for an editable install is? Is it `poetry shell` followed by `poetry install`?

This gives
```
(auto-pi-lot-JL3WMapu-py3.9) mouse@jellyfish:~/dev/autopilot$ poetry install
Installing dependencies from lock file
Warning: The lock file is not up to date with the latest changes in pyproject.toml. You may be getting outdated dependencies. Run update to update them.
```
followed by a permission error

thanks for any tips!

sneakers-the-rat, 2022-08-02:
Poetry does support editable installs with pip now! It might be a pip version thing if it's not working, I remember that happened somewhat recently. The `pyproject.toml` is intended to fully replace `setup.py`

my bad on the outdated lockfile. I'm not sure about the permissions error, could you copy/paste that?

cxrodgers, 2022-08-03:
Oh great! I updated pip to 22.2 and now it works well, that was an easy fix. Weird because the top hits for "editable install with poetry" was just giving me people arguing on github issue pages.

cxrodgers, 2022-08-03:
Oh, one thing though, it `pip install -e .` seems to install everything properly other than `rich`. I still have to do `pip install rich` manually. I don't know enough about toml files to understand why this is the case.

sneakers-the-rat, 2022-08-04:
I also have no idea why it fails to install rich , as it's in both the pyproject.toml file: https://github.com/auto-pi-lot/autopilot/blob/04b5968ba02c8a1413a27eb6a138b6a186b130f1/pyproject.toml#L51 and the lockfile. It could be since it is listed as an extra for the docs and tests that it becomes optional generally? But I added it to docs and tests because it wasn't installing there either... I'll raise an issue with them.