auto-pi-lot/autopilot #117: LEDS RGB can't be set or pull

mnmneE, 2021-08-02:
Hello
 
I am trying to implement a simplified version of Freewater task using LEDS RGB and pushbuttons only by developping the Blink task example, I want to trigger the push button (Digital_IN) object to change the color of the led,
but the pilot terminal throws always KEY ERROR related to the names of the leds rgb pins;
please find attached screen shots for both json file and the script of the blink task with the errors followed in a separate comment

Regards

mnmneE, 2021-08-02:
[image]
error with Pulse method

mnmneE, 2021-08-02:
[image]
error with set_LEDS METHOD

sneakers-the-rat, 2021-08-02:
It looks like the LEDs are configured only with one pin each, but the `LED_RGB` class needs a list of three pins (r, g, b) to work with. see the docs: https://docs.auto-pi-lot.com/en/latest/autopilot.hardware.gpio.html#autopilot.hardware.gpio.LED_RGB

It should have been the case that an error message is raised telling you that, but it looks like I wrote it incorrectly, where the `ValueError` is created but never raised (this is an old misunderstanding I had about how exceptions worked that is being corrected throughougt the package): 
https://github.com/wehr-lab/autopilot/blob/7617e26d1c4ec4273c44f3e9a7f314fbfd23cec1/autopilot/hardware/gpio.py#L994

The argument should also be `pins` rather than `pin` -- I imagine this is a result of the hardware setup wizard not filtering arguments from parent classes (eg. it see that `LED_RGB` asks for `pins`, but the `GPIO` class asks for `pin` and there isn't a system in place to disambiguate those yet). 

It looks like you have three LEDs configured -- are those three separate LEDs? or are thsoe the r, g, b, channels of one LED? 

From there, since the object would be created without creating the relevant `PWM` objects in the `channels` dictionary (see here: https://github.com/wehr-lab/autopilot/blob/7617e26d1c4ec4273c44f3e9a7f314fbfd23cec1/autopilot/hardware/gpio.py#L985-L992 ) then all the further calls will throw errors as they attempt to use them (those `KeyError` s)

Additionally it looks like you're using the `set_leds` method -- that method rather inelegantly requires that you name the group that the hardware is in `LEDS` , so you would change `'GPIO'` to `'LEDS'` in `prefs.json` and the task's `HARDWARE` dictionary.  This sort of brittle handling of naming of hardware objects is a longstanding problem, and is one that is now on the way to being solved -- the registry system that I'm pulling today or tomorrow makes it much easier to work with abstract references to hardware classes, so it should be much less necessary to have these very specific names for hardware groups. 

It also looks like `pull` is an empty string, which should just cause a warning. The `LED_RGB` class uses the `polarity` argument to set both the pull direction and the direction of the logic (high = on vs. low = on).

sneakers-the-rat, 2021-08-02:
added ValueErrors in the class that should make this behavior clearer:
https://github.com/wehr-lab/autopilot/commit/8f8d9abe0f8b2665068d0eae73ff29ef2c506b36

will pull to main with v0.4.0

mnmneE, 2021-08-02:
Thanks alot for your answers, and sorry for the quality of screenshots, I was about to modify them but your response was faster.

sneakers-the-rat, 2021-08-02:
I like to think of starting with whatever information is given in the initial issue as 'offroad debugging' :). let me know if that works, or if any further errors come up! We can keep this issue open until you're sure the problem is resolved.

mnmneE, 2021-08-05:
Thanks again for your help, I generated the hardware JSON file from the terminal and directly noticed the place where R; G;B channels must be configured; and so my issue is solved.

closed 2021-08-06 by sneakers-the-rat