auto-pi-lot/autopilot #142: PiCamera instance variables

skeltoh, 2021-10-29:
The example code for capturing video here (https://docs.auto-pi-lot.com/en/latest/guide/quickstart.html) produces an error (I added the name to avoid a warning):

> from autopilot.hardware.cameras import PiCamera
> cam = PiCamera(name='cam')
> cam.write('test_video.mp4')
> cam.capture(timed=10)

> ---------------------------------------------------------------------------
> AttributeError                            Traceback (most recent call last)
> /tmp/ipykernel_14379/453994599.py in <module>
> ----> 1 cam.write('test_video.mp4')
> 
> ~/git/autopilot/autopilot/hardware/cameras.py in write(self, output_filename, timestamps, blosc)
>     402         self.blosc = blosc
>     403         self._write_q = mp.Queue()
> --> 404         self.writer = Video_Writer(self._write_q, output_filename, self.fps, timestamps=timestamps, blosc=blosc)
>     405         self.writer.start()
>     406         self.writing.set()
> 
> ~/git/autopilot/autopilot/hardware/cameras.py in fps(self)
>     724             int - fps
>     725         """
> --> 726         return self._fps
>     727 
>     728     @fps.setter
> 
> AttributeError: 'PiCamera' object has no attribute '_fps'

If I change the code to 

> from autopilot.hardware.cameras import PiCamera
> cam = PiCamera(name='cam')
> cam._fps = 30
> cam.write('test_video.mp4')
> cam.capture(timed=10)

Then I get:

> Exception in thread Thread-8:
> Traceback (most recent call last):
>   File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
>     self.run()
>   File "/usr/lib/python3.7/threading.py", line 865, in run
>     self._target(*self._args, **self._kwargs)
>   File "/home/pi/git/autopilot/autopilot/hardware/cameras.py", line 226, in _capture
>     self.capture_init()
>   File "/home/pi/git/autopilot/autopilot/hardware/cameras.py", line 788, in capture_init
>     self._picam_writer = self.PiCamera_Writer(self.resolution, self.format)
> AttributeError: 'PiCamera' object has no attribute 'format'

If I add:

> cam.format = 'rgb'

Then I get:

> Exception in thread Thread-8:
> Traceback (most recent call last):
>   File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
>     self.run()
>   File "/usr/lib/python3.7/threading.py", line 865, in run
>     self._target(*self._args, **self._kwargs)
>   File "/home/pi/git/autopilot/autopilot/hardware/cameras.py", line 226, in _capture
>     self.capture_init()
>   File "/home/pi/git/autopilot/autopilot/hardware/cameras.py", line 788, in capture_init
>     self._picam_writer = self.PiCamera_Writer(self.resolution, self.format)
>   File "/home/pi/git/autopilot/autopilot/hardware/cameras.py", line 847, in __init__
>     self.resolution[0]+ 31 // 32 * 32,
> TypeError: 'NoneType' object is not subscriptable

If I add:

> cam.resolution = (640,480)

Then I get:

> ---------------------------------------------------------------------------
> AttributeError                            Traceback (most recent call last)
> /tmp/ipykernel_16972/3992256397.py in <module>
>       2 cam._fps = 30
>       3 cam.format = 'rgb'
> ----> 4 cam.resolution = (640,480)
> 
> ~/git/autopilot/autopilot/hardware/cameras.py in resolution(self, resolution)
>     706     def resolution(self, resolution: typing.Tuple[int, int]):
>     707         self._resolution = resolution
> --> 708         if self._picam_writer is not None:
>     709             self._picam_writer.resolution = self._resolution
>     710         if self.initialized.is_set() and not self.capturing.is_set():
> 
> AttributeError: 'PiCamera' object has no attribute '_picam_writer'

At this point, it seems like something strange is going on with getters/setter methods in this class. It seems kind of odd since I see `self._picam_writer = None` in PiCamera.__init__.

As a note, I actually had another error before I installed ffmpeg with apt. Maybe that should be added to the setup scripts.

sneakers-the-rat, 2021-10-29:
I think I see what's going on here -- 

the camera module checks if the picamera package is present:
https://github.com/wehr-lab/autopilot/blob/d352feef3b8aac8699fbc4c27270580d0a48ab0f/autopilot/hardware/cameras.py#L37-L41

and then if it isn't detected, the PiCamera class bails on initializing:
https://github.com/wehr-lab/autopilot/blob/d352feef3b8aac8699fbc4c27270580d0a48ab0f/autopilot/hardware/cameras.py#L655-L657

and an exception is *logged* but not *raised*. 

So all this that follows & sets up the object doesn't happen, but the object is still created: 
https://github.com/wehr-lab/autopilot/blob/d352feef3b8aac8699fbc4c27270580d0a48ab0f/autopilot/hardware/cameras.py#L659-L668

Which would explain all the above behavior.

does this fix? https://github.com/wehr-lab/autopilot/commit/9637ec771f4ce181a6b5139a0f8d67eb7d341803

skeltoh, 2021-11-01:
Looks like the real problem was I didn't have PiCamera installed, but that bug fix alerts if it isn't. Thanks!

closed 2021-11-01 by skeltoh