auto-pi-lot/autopilot #151: doc error on autopilot.setup command

cxrodgers, 2022-02-11:
Setting up a Terminal for the first time in a while and I think (??) there is a bug in the doc:

https://docs.auto-pi-lot.com/en/latest/guide/installation.html

```
To start the guided process, run the following line.
python3 -m autopilot.setup
```

The command as written gives me an error. Shouldn't this be `python3 - m autopilot.setup.setup_autopilot`? 

If so, I'm happy to make the PR. Just wanted to first make sure I'm not missing something obvious.

sneakers-the-rat, 2022-02-11:
as of v0.4.0 `autopilot.setup` has a `__main__.py` module that runs the setup!
https://github.com/wehr-lab/autopilot/blob/main/autopilot/setup/__main__.py

so these are  both valid:

* `python -m autopilot.setup`
https://github.com/wehr-lab/autopilot/blob/e408c08e76df8d8c930edfb4ba58e31cd1ec4d87/autopilot/setup/__main__.py#L1-L3
* `python -m autopilot.setup.setup_autopilot`
https://github.com/wehr-lab/autopilot/blob/e408c08e76df8d8c930edfb4ba58e31cd1ec4d87/autopilot/setup/setup_autopilot.py#L399-L400

If either doesn't work though, that's a bug!

Re:
https://github.com/wehr-lab/autopilot/pull/115
https://github.com/wehr-lab/autopilot/issues/33
https://github.com/wehr-lab/autopilot/commit/ccec973f68e83ed4320fb7addc295a9117583bc5

cxrodgers, 2022-02-11:
Oh shoot I know what happened, I'm in my own fork and I haven't caught up with the most recent version. That explains it, sorry!

closed 2022-02-11 by cxrodgers