auto-pi-lot/autopilot #154: jack failure

mikewehr, 2022-02-18:
pi@raspberrypi:~ $ sudo reboot
Connection to 192.168.0.197 closed by remote host.
Connection to 192.168.0.197 closed.
wehr@wehrterminal2:~$ ssh pi@192.168.0.197 
pi@192.168.0.197's password: 
Linux raspberrypi 5.10.63-v7+ #1496 SMP Wed Dec 1 15:58:11 GMT 2021 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Fri Feb 18 23:06:14 2022 from 192.168.0.147
pi@raspberrypi:~ $ ps -aux
USER       PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
root         1 16.4  0.8  15144  7644 ?        Ss   23:06   0:05 /sbin/init
root         2  0.0  0.0      0     0 ?        S    23:06   0:00 [kthreadd]
root         3  0.0  0.0      0     0 ?        I<   23:06   0:00 [rcu_gp]
root         4  0.0  0.0      0     0 ?        I<   23:06   0:00 [rcu_par_gp]
root         5  0.0  0.0      0     0 ?        I    23:06   0:00 [kworker/0:0-events_long]
root         6  0.0  0.0      0     0 ?        I<   23:06   0:00 [kworker/0:0H-kblockd]
root         7  0.1  0.0      0     0 ?        R    23:06   0:00 [kworker/u8:0-brcmf_wq/mmc1:0001:1]
root         8  0.0  0.0      0     0 ?        I<   23:06   0:00 [mm_percpu_wq]
root         9  0.0  0.0      0     0 ?        S    23:06   0:00 [rcu_tasks_rude_]
root        10  0.0  0.0      0     0 ?        S    23:06   0:00 [rcu_tasks_trace]
root        11  0.2  0.0      0     0 ?        S    23:06   0:00 [ksoftirqd/0]
root        12  0.2  0.0      0     0 ?        I    23:06   0:00 [rcu_sched]
root        13  0.0  0.0      0     0 ?        S    23:06   0:00 [migration/0]
root        14  0.0  0.0      0     0 ?        S    23:06   0:00 [cpuhp/0]
root        15  0.0  0.0      0     0 ?        S    23:06   0:00 [cpuhp/1]
root        16  0.0  0.0      0     0 ?        S    23:06   0:00 [migration/1]
root        17  0.1  0.0      0     0 ?        S    23:06   0:00 [ksoftirqd/1]
root        18  0.0  0.0      0     0 ?        I    23:06   0:00 [kworker/1:0-events_long]
root        19  0.0  0.0      0     0 ?        I<   23:06   0:00 [kworker/1:0H-events_highpri]
root        20  0.0  0.0      0     0 ?        S    23:06   0:00 [cpuhp/2]
root        21  0.0  0.0      0     0 ?        S    23:06   0:00 [migration/2]
root        22  0.1  0.0      0     0 ?        S    23:06   0:00 [ksoftirqd/2]
root        23  0.0  0.0      0     0 ?        I    23:06   0:00 [kworker/2:0-cgroup_destroy]
root        24  0.0  0.0      0     0 ?        I<   23:06   0:00 [kworker/2:0H-kblockd]
root        25  0.0  0.0      0     0 ?        S    23:06   0:00 [cpuhp/3]
root        26  0.0  0.0      0     0 ?        S    23:06   0:00 [migration/3]
root        27  0.3  0.0      0     0 ?        S    23:06   0:00 [ksoftirqd/3]
root        28  0.0  0.0      0     0 ?        I    23:06   0:00 [kworker/3:0-events]
root        29  0.0  0.0      0     0 ?        I<   23:06   0:00 [kworker/3:0H-kblockd]
root        30  0.0  0.0      0     0 ?        S    23:06   0:00 [kdevtmpfs]
root        31  0.0  0.0      0     0 ?        I<   23:06   0:00 [netns]
root        32  0.0  0.0      0     0 ?        I    23:06   0:00 [kworker/0:1-rcu_gp]
root        33  0.0  0.0      0     0 ?        I    23:06   0:00 [kworker/1:1-events_long]
root        34  0.0  0.0      0     0 ?        S    23:06   0:00 [kauditd]
root        35  0.2  0.0      0     0 ?        I    23:06   0:00 [kworker/0:2-events]
root        36  0.0  0.0      0     0 ?        S    23:06   0:00 [khungtaskd]
root        37  0.0  0.0      0     0 ?        S    23:06   0:00 [oom_reaper]
root        38  0.0  0.0      0     0 ?        I<   23:06   0:00 [writeback]
root        39  0.0  0.0      0     0 ?        S    23:06   0:00 [kcompactd0]
root        57  0.0  0.0      0     0 ?        I<   23:06   0:00 [kblockd]
root        58  0.0  0.0      0     0 ?        I<   23:06   0:00 [blkcg_punt_bio]
root        59  0.0  0.0      0     0 ?        S    23:06   0:00 [watchdogd]
root        60  0.0  0.0      0     0 ?        I    23:06   0:00 [kworker/2:1-events]
root        61  0.0  0.0      0     0 ?        I<   23:06   0:00 [kworker/1:1H-kblockd]
root        62  0.0  0.0      0     0 ?        I<   23:06   0:00 [rpciod]
root        63  0.0  0.0      0     0 ?        I<   23:06   0:00 [kworker/u9:0-hci0]
root        64  0.0  0.0      0     0 ?        I<   23:06   0:00 [xprtiod]
root        65  0.0  0.0      0     0 ?        S    23:06   0:00 [kswapd0]
root        66  0.0  0.0      0     0 ?        I<   23:06   0:00 [nfsiod]
root        67  0.3  0.0      0     0 ?        I    23:06   0:00 [kworker/3:1-events]
root        68  0.4  0.0      0     0 ?        I    23:06   0:00 [kworker/1:2-events]
root        69  0.0  0.0      0     0 ?        I<   23:06   0:00 [iscsi_eh]
root        70  0.0  0.0      0     0 ?        I<   23:06   0:00 [iscsi_destroy]
root        71  0.0  0.0      0     0 ?        I<   23:06   0:00 [dwc_otg]
[... 130 lines cut ...]
Cannot connect to server request channel
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/pi/git/autopilot/autopilot/core/pilot.py", line 817, in <module>
    a = Pilot()
  File "/home/pi/git/autopilot/autopilot/core/pilot.py", line 181, in __init__
    self.init_audio()
  File "/home/pi/git/autopilot/autopilot/core/pilot.py", line 651, in init_audio
    self.server = jackclient.JackClient()
  File "/home/pi/git/autopilot/autopilot/stim/sound/jackclient.py", line 158, in __init__
    self.client = jack.Client(self.name)
  File "/home/pi/.venv/autopilot/lib/python3.7/site-packages/jack.py", line 213, in __init__
    raise JackOpenError(name, self._status)
jack.JackOpenError: Error initializing "jack_client": <jack.Status 0x11: failure, server_failed>
pi@raspberrypi:~ $

sneakers-the-rat, 2022-02-20:
looks like there's already an autopilot process running. is the systemd daemon enabled? try `sudo systemctl status autopilot` or else there is another shell open with it running? in any case the error is that the sound card has already been reserved, presumably by that running jackd process, but this also happens when PulseAudio is running. it also could be a memory issue, as it says it can't lock down the memory it needs. can you do a `cat /etc/security/limits.conf` ?

mikewehr, 2022-02-21:
sudo systemctl stop autopilot
sudo systemctl disable autopilot
solved this issue for me

closed 2022-02-21 by mikewehr