auto-pi-lot/autopilot #164: libjackserver.so not found bc it is in /usr/lib/arm-linux-gnueabihf

Rodgers-PAC-Lab, 2022-03-16:
@Rowanon and I are installing autopilot on a fresh raspberry pi and running into some issues with jack, which I don't _think_ I had with previous versions of the OS. 

Basically when we start jackd we get:

> pi@rpi09:~ $ jackd -P75 -p16 -t2000 -dalsa -dhw:sndrpihifiberry -P -n3 -s -r192000
> jackd: error while loading shared libraries: libjackserver.so.0: cannot open shared object file: No such file or directory

`libjackserver.so.0` does exist in directory `/usr/lib/arm-linux-gnueabihf`, I think because that's where we told `waf` to put it during the "install jackd from source" option in autopilot setup. 

This is a fix:
`export LD_LIBRARY_PATH=/usr/lib/arm-linux-gnueabihf`

For now this fix is fine, but I wonder, is the new version of Raspi Lite OS looking for these libs in a different place, and if so, should we update the way we tell waf to install them?

sneakers-the-rat, 2022-03-16:
ok yes glad you got a fix to work. there are some new oddities in the way that raspios bullseye does sound that I haven't worked out exactly yet (I just added a note to the docs to that effect the other day: https://docs.auto-pi-lot.com/en/latest/stim/sound/jackclient.html )

that sounds like something the linker should have taken care of, but until we figure out a more permanent fix we can add that to .bashrc or something. 

but ya widespread issues with Jack in particular, not at my desk atm but I was sort of waiting for the dust to settle first. it has something to do with switching up they way they handle PulseAudio, but still trying to get the lay of the land

cxrodgers, 2022-03-17:
sounds good, thanks jonny! We'll add `export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/lib/arm-linux-gnueabihf` to the end of `~/.bashrc` to make this fix "permanent".

Also, sorry, I know this is a serious venue for Autopilot discussion but hearing about pulse audio problems again reminds me of this old linux joke:
[image]

sneakers-the-rat, 2022-03-20:
OK! Keeping this open.

> I know this is a serious venue for Autopilot discussion 

i will have you know i take shitposting very seriously.