auto-pi-lot/autopilot #183: Cannot create subject

cxrodgers, 2022-08-02:
I am on the `dev` branch, so v0.5. I'm having trouble creating a subject by clicking the "+" in the Terminal GUI under the green button. 

If I complete the subject's "id" textbox, but none of the other optional fields, and click "OK" I get this:

```
[22-08-02T16:12:38] DEBUG    [gui.widgets.subject.New_Subject_Wizard] Clicked OK to create subject                                                                  subject.py:57
Traceback (most recent call last):
  File "/home/mouse/dev/autopilot/autopilot/gui/widgets/subject.py", line 58, in _accept
    model = self.bio_tab.validate(dialog=True)
  File "/home/mouse/dev/autopilot/autopilot/gui/widgets/model.py", line 232, in validate
    kwargs = self.dict()
  File "/home/mouse/dev/autopilot/autopilot/gui/widgets/model.py", line 138, in dict
    kwargs[key] = input.value()
  File "/home/mouse/dev/autopilot/autopilot/gui/widgets/input.py", line 271, in value
    return literal_eval(self._widget.text())
  File "/home/mouse/miniconda3/envs/autopilot/lib/python3.9/ast.py", line 62, in literal_eval
    node_or_string = parse(node_or_string, mode='eval')
  File "/home/mouse/miniconda3/envs/autopilot/lib/python3.9/ast.py", line 50, in parse
    return compile(source, filename, mode, flags,
  File "<unknown>", line 0
    
SyntaxError: unexpected EOF while parsing
```
If I put text in all the optional fields in the bio, I get this

```
[22-08-02T16:13:10] DEBUG    [gui.widgets.subject.New_Subject_Wizard] Clicked OK to create subject                                                                  subject.py:57
Traceback (most recent call last):
  File "/home/mouse/dev/autopilot/autopilot/gui/widgets/subject.py", line 58, in _accept
    model = self.bio_tab.validate(dialog=True)
  File "/home/mouse/dev/autopilot/autopilot/gui/widgets/model.py", line 232, in validate
    kwargs = self.dict()
  File "/home/mouse/dev/autopilot/autopilot/gui/widgets/model.py", line 138, in dict
    kwargs[key] = input.value()
  File "/home/mouse/dev/autopilot/autopilot/gui/widgets/input.py", line 271, in value
    return literal_eval(self._widget.text())
  File "/home/mouse/miniconda3/envs/autopilot/lib/python3.9/ast.py", line 105, in literal_eval
    return _convert(node_or_string)
  File "/home/mouse/miniconda3/envs/autopilot/lib/python3.9/ast.py", line 104, in _convert
    return _convert_signed_num(node)
  File "/home/mouse/miniconda3/envs/autopilot/lib/python3.9/ast.py", line 78, in _convert_signed_num
    return _convert_num(node)
  File "/home/mouse/miniconda3/envs/autopilot/lib/python3.9/ast.py", line 69, in _convert_num
    _raise_malformed_node(node)
  File "/home/mouse/miniconda3/envs/autopilot/lib/python3.9/ast.py", line 66, in _raise_malformed_node
    raise ValueError(f'malformed node or string: {node!r}')
ValueError: malformed node or string: <ast.Name object at 0x7f1044048df0>
```

sneakers-the-rat, 2022-08-02:
What the heck i'm also getting this, but am sure that this worked because there's a roundtrip test of getting and setting from the widget here: https://github.com/auto-pi-lot/autopilot/blob/04b5968ba02c8a1413a27eb6a138b6a186b130f1/tests/test_gui/test_widget_model.py#L41

In any case i'm going to make a more informative error message when validating (there already is one, but this happens before it: https://github.com/auto-pi-lot/autopilot/blob/04b5968ba02c8a1413a27eb6a138b6a186b130f1/autopilot/gui/widgets/model.py#L234-L243

1 second

cxrodgers, 2022-08-02:
We need a test for the test! :laughing:

sneakers-the-rat, 2022-08-02:
OK got it:

https://github.com/auto-pi-lot/autopilot/commit/0a2f081369b794920e62c1afee9e8cbf2e681dd6

see the diff there for specifics, but in short:

- Models can be recursive (ie. have other models as fields), and those submodels can be optional
- When getting values from optional models, the test for whether they were enabled was backwards (oops) so it was trying to get values from models even when they were empty/inactive
- When fields are empty, the input widgets weren't correctly handling that - so `ast.literal_eval` chokes on an empty string
- Some of the fields expect particular input - eg. the tags field is supposed to receive a dict, but at the moment that's not clearly specified in the visual appearance of the widget (there are tooltips but you have to hover over the field name, which isnt' obvious. intend to fix that),  but there wasn't a specific error message telling you that.
- The tests only tested a complete biography, not an incomplete one, so this wasn't covered - added a separate test just to make sure that happens, though more tests for the individual input widgets are needed, but i have to do this defense rn lol.

lmk if that doesn't work.

cxrodgers, 2022-08-03:
Works great, thanks!