auto-pi-lot/autopilot #186: init_hardware on a Child fails to start pigpiod due to threading problem

cxrodgers, 2022-08-04:
Having trouble initializing hardware on a Child Pi. As far as I can tell, the first time a GPIO is initalized, it creates a subprocess for pigpiod and stores it in a global variable PIGPIO_DAEMON, unless that global already exists, in which case it uses the existing value. Problem occurs because for some reason the `signal` line can't be executed in any thread except the main one, and I guess the Child task is being run in a thread? I do not get this problem on the Parent Pi. 

Seems like the fix is to call external.start_pigpiod() in the main thread on the Child, if I can figure out how to do that.

```
                    DEBUG    [hardware.gpio.Digital_In.POKES_L] Logger created: hardware.gpio.Digital_In.POKES_L                                                   loggers.py:164
                    ERROR    [tasks.paft_child.PAFT_Child.rpi01] Pin could not be instantiated - Type: POKES, Pin: L                                            paft_child.py:448
                             Got exception:signal only works in main thread of the main interpreter                                                                              
                             ╭────────────────────────────────────────────── Traceback (most recent call last) ───────────────────────────────────────────────╮                  
                             │ /home/pi/dev/autopilot/autopilot/tasks/paft_child.py:427 in init_hardware                                                      │                  
                             │                                                                                                                                │                  
                             │   424 │   │   │   │   │   if isinstance(hw_args, dict):                                                                        │                  
                             │   425 │   │   │   │   │   │   if 'name' not in hw_args.keys():                                                                 │                  
                             │   426 │   │   │   │   │   │   │   hw_args['name'] = "{}_{}".format(type, pin)                                                  │                  
                             │ ❱ 427 │   │   │   │   │   │   hw = handler(**hw_args)                                                                          │                  
                             │   428 │   │   │   │   │   else:                                                                                                │                  
                             │   429 │   │   │   │   │   │   hw_name = "{}_{}".format(type, pin)                                                              │                  
                             │   430 │   │   │   │   │   │   hw = handler(hw_args, name=hw_name)                                                              │                  
                             │                                                                                                                                │                  
                             │ /home/pi/dev/autopilot/autopilot/hardware/gpio.py:763 in __init__                                                              │                  
                             │                                                                                                                                │                  
                             │    760 │   │   """                                                                                                             │                  
                             │    761 │   │                                                                                                                   │                  
                             │    762 │   │   """                                                                                                             │                  
                             │ ❱  763 │   │   super(Digital_In, self).__init__(pin, **kwargs)                                                                 │                  
                             │    764 │   │                                                                                                                   │                  
                             │    765 │   │   # pull the resistor in the off direction and set the trigger to be the on                                       │                  
                             │        direction                                                                                                               │                  
                             │    766 │   │   self.pull = self.off                                                                                            │                  
                             │                                                                                                                                │                  
                             │ /home/pi/dev/autopilot/autopilot/hardware/gpio.py:189 in __init__                                                              │                  
                             │                                                                                                                                │                  
                             │    186 │   │                                                                                                                   │                  
                             │    187 │   │   # init pigpio                                                                                                   │                  
                             │    188 │   │   self.CONNECTED = False                                                                                          │                  
                             │ ❱  189 │   │   self.CONNECTED = self.init_pigpio()                                                                             │                  
                             │    190 │   │                                                                                                                   │                  
                             │    191 │   │   # set default attributes                                                                                        │                  
                             │    192                                                                                                                         │                  
                             │                                                                                                                                │                  
                             │ /home/pi/dev/autopilot/autopilot/hardware/gpio.py:208 in init_pigpio                                                           │                  
                             │                                                                                                                                │                  
                             │    205 │   │   Returns:                                                                                                        │                  
                             │    206 │   │   │   bool: True if connection was successful, False otherwise                                                    │                  
                             │    207 │   │   """                                                                                                             │                  
                             │ ❱  208 │   │   self.pigpiod = external.start_pigpiod()                                                                         │                  
                             │    209 │   │   self.pig = pigpio.pi()                                                                                          │                  
                             │    210 │   │   if self.pig.connected:                                                                                          │                  
                             │    211 │   │   │   return True                                                                                                 │                  
[... cut ...]
                             │   49                                                                                                                           │                  
                             │   50                                                                                                                           │                  
                             ╰────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────╯                  
                             ValueError: signal only works in main thread of the main interpreter  
```

cxrodgers, 2022-08-04:
Ok I think I found the problem.
* pigpiod is supposed to be started by this line: https://github.com/auto-pi-lot/autopilot/blob/04b5968ba02c8a1413a27eb6a138b6a186b130f1/autopilot/agents/pilot.py#L185
* But that line checks for a pref called PIGPIOD
* The only documented similar pref is called PIGPIO https://github.com/auto-pi-lot/autopilot/blob/04b5968ba02c8a1413a27eb6a138b6a186b130f1/autopilot/prefs.py#L336
* There are no other references to either of these prefs anywhere else
* On v0.4, I didn't have either of those prefs set, and I guess it was defaulting to true. For some reason since the upgrade to v0.5, I now have a variable PIGPIO which is "false", though I think the real problem was the _lack_ of a pref called PIGPIOD which is true.
* This wasn't a problem on my Parent Pi because it doesn't have any hardware, but it may also thread things differently and/or initialize pigpiod somewhere else.

Workaround: add a pref called PIGPIOD and set it to "true".
Fix (TODO): consistently call this variable either PIGPIO or PIGPIOD. Also check whether autopilot.setup is setting it to false (I probably forgot to check this box during setup, but it probably also should default to true if the user forgets)

sneakers-the-rat, 2022-08-04:
YES. that is extremely annoying. Making Prefs explicit is the thing I didn't have time to get to when switching over to formal data models for v0.5.0. The way it works now is bad for exactly this reason (not clearly documented, some prefs are implicitly accessed, etc.)

related to : https://github.com/auto-pi-lot/autopilot/issues/155

Launching pigpiod as a byproduct of initializing a GPIO object is for the purpose of being able to use them outside of a task/agent context, which was a requested (and useful) feature, but pigpio is just a pretty fussy program generally and so yes we need a better way of killing the process. A better way of launching it than is done now (a function call with an implicit global module state) would be to have explicit process handling classes, just haven't had the time to implement yet.