auto-pi-lot/autopilot #194: installation error

lucasogit, 2022-10-21:
I get this error when trying to install auto-pi-lot on ubuntu 22.04.1 LTS using pip install auto-pi-lot: 

`lucaso@lucaso-Surface-Book:~$ pip install auto-pi-lot
Defaulting to user installation because normal site-packages is not writeable
Collecting auto-pi-lot
  Using cached auto-pi-lot-0.4.4.tar.gz (268 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error
  
  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [20 lines of output]
      Traceback (most recent call last):
        File "/usr/lib/python3/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 363, in <module>
          main()
        File "/usr/lib/python3/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 345, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
        File "/usr/lib/python3/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 130, in get_requires_for_build_wheel
          return hook(config_settings)
        File "/usr/lib/python3/dist-packages/setuptools/build_meta.py", line 162, in get_requires_for_build_wheel
          return self._get_build_requires(
        File "/usr/lib/python3/dist-packages/setuptools/build_meta.py", line 143, in _get_build_requires
          self.run_setup()
        File "/usr/lib/python3/dist-packages/setuptools/build_meta.py", line 267, in run_setup
          super(_BuildMetaLegacyBackend,
        File "/usr/lib/python3/dist-packages/setuptools/build_meta.py", line 158, in run_setup
          exec(compile(code, __file__, 'exec'), locals())
        File "setup.py", line 34, in <module>
          REQUIREMENTS = load_requirements('requirements.txt')
        File "setup.py", line 20, in load_requirements
          with open(req_file, 'r') as f:
      FileNotFoundError: [Errno 2] No such file or directory: 'requirements.txt'
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.`

cxrodgers, 2022-10-21:
Have you tried upgrading to most recent version of pip? I don't know if this is related or not, but I had problems installing with pip after the installation changed to poetry-type requirements files. Upgraded my pip and then it was all good

lucasogit, 2022-10-21:
Just tried that and still getting errors.

sneakers-the-rat, 2022-10-22:
*annoying* and *embarassing* it's because of the arcane setuptools shit.

i removed the wildcard from the `MANIFEST.in` line that includes the requirements, so the ye olde setup.py script that chooses between different requirements files instead of actually having reasonable dependency management chokes when it's missing. will issue a patch now

sneakers-the-rat, 2022-10-22:
fixed: https://pypi.org/project/auto-pi-lot/0.4.4.post0/
https://github.com/auto-pi-lot/autopilot/tree/v0.4.4.post0
dang the packaging here is such a mess. thank god for v0.5.0

closed 2022-10-22 by sneakers-the-rat