auto-pi-lot/autopilot #195: installation issue the sequel

lucasogit, 2022-10-26:
I get this error when installing with pip3 install auto-pi-lot:

Building wheels for collected packages: blosc
  Building wheel for blosc (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building wheel for blosc (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [9 lines of output]
        File "/tmp/pip-build-env-ayfnen4r/overlay/local/lib/python3.10/dist-packages/skbuild/setuptools_wrap.py", line 613, in setup
          cmkr = cmaker.CMaker(cmake_executable)
        File "/tmp/pip-build-env-ayfnen4r/overlay/local/lib/python3.10/dist-packages/skbuild/cmaker.py", line 141, in __init__
          self.cmake_version = get_cmake_version(self.cmake_executable)
        File "/tmp/pip-build-env-ayfnen4r/overlay/local/lib/python3.10/dist-packages/skbuild/cmaker.py", line 95, in get_cmake_version
          raise SKBuildError(
      Traceback (most recent call last):
      
      Problem with the CMake installation, aborting build. CMake executable is cmake
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for blosc
Failed to build blosc
ERROR: Could not build wheels for blosc, which is required to install pyproject.toml-based projects

sneakers-the-rat, 2022-11-01:
that looks like a blosc problem but i'll investigate.

sneakers-the-rat, 2023-05-23:
Yes - sorry i have been having a difficult few months and slow to respond to issues. this was an issue with not having cmake installed if i recall correctly, so that would be the first thing to try:
```
apt update && apt install -y cmake
```

Alternatively with many python build problems I have found the problem could come from having an old version of pip. Finally you could try installing the blosc system-level libraries.

I am still in the process of moving labs, which has been dragged out due to unforeseen circumstances, and don't have my raspi cluster set up yet to test, but please let me know if those don't resolve and I will hopefully have them back up by then.