auto-pi-lot/autopilot #206: Error when trying to add a new pilot on terminal

lucasogit, 2024-01-24:
I am messing around with v0.5 and am getting a decent amount of PySide6 errors in the terminal GUI. Here is one specific to trying to add a new pilot:

```
Traceback (most recent call last):
  File "/home/lab/autopilot/autopilot/utils/invoker.py", line 42, in event
    event.fn(*event.args, **event.kwargs)
  File "/home/lab/autopilot/autopilot/gui/widgets/terminal.py", line 114, in add_pilot
    pilot_panel = Pilot_Panel(pilot_id, subject_list, self.start_fn, self.ping_fn, self.create_subject)
  File "/home/lab/autopilot/autopilot/gui/widgets/terminal.py", line 294, in init
    self.init_ui()
  File "/home/lab/autopilot/autopilot/gui/widgets/terminal.py", line 302, in init_ui
    label = QtWidgets.QLabel(self.pilot)
TypeError: 'PySide6.QtWidgets.QLabel.init' called with wrong argument types:
  PySide6.QtWidgets.QLabel.init(bool)
Supported signatures:
  PySide6.QtWidgets.QLabel.init(Optional[PySide6.QtWidgets.QWidget] = None, PySide6.QtCore.Qt.WindowType = Default(Qt.WindowFlags))
  PySide6.QtWidgets.QLabel.init(str, Optional[PySide6.QtWidgets.QWidget] = None, PySide6.QtCore.Qt.WindowType = Default(Qt.WindowFlags))

```

sneakers-the-rat, 2024-01-24:
Oop, didnt think anyone was still out there using this thing. Admittedly it was a sloppy release mostly to leave it in a non-partially versioned state. Ill fix this and write tests tmrw

sneakers-the-rat, 2024-01-25:
hmm how did a bool end up there

sneakers-the-rat, 2024-01-26:
OK this particular bug is fixed in https://github.com/auto-pi-lot/autopilot/commit/fa596b68d5572cb1bafdc7bedfa374250659af80

but we are still getting PySide6 bugs, specifically a dreaded threading problem throwing a segfault exactly here: https://github.com/auto-pi-lot/autopilot/blob/fa596b68d5572cb1bafdc7bedfa374250659af80/autopilot/gui/widgets/subject.py#L131

I worked through that for several hours and can't really make sense of it. Stuff that is true:
- That exact spot succeeds when the widget is first created
- It seems to be only when it is triggered by the event that the segfault happens
- The traceback shows a bunch of thread errors from the networking nodes, but removing all of them from instantiation (within Terminal and plot widgets) doesn't fix the problem
- the input (a list of strings) is fine

at the moment i'm like:
????

cxrodgers, 2024-01-26:
I ran into some segfaults at some point. Sometimes there would be these bizarre symptoms where the culprit would appear to be one innocuous line, but inserting debug statements would subtly change the behavior. I had the feeling that some of the issue was multiple threads / objects opening the HDF5 file. At the time, you suggested the plotting thread was a likely culprit, but I never managed to test that. By now my fork is so heavily modified that it's not a useful point of comparison. Sorry I can't be helpful...

sneakers-the-rat, 2024-01-26:
I am pretty sure this is the long arc of "you actually shouldn't use `threading` with Qt" coming back to bite us. What's especially odd is *exactly* where the bug happens - there are similar operations that happen all over the place, even within the same method, and that same line works in a prior call! but for some reason it is reliably *that one line* called in *exactly that way* that triggers a segfault.

I might just rollback all the PySide6 changes (curse me for not being disciplined with my upgrade, i was just trying to do some light dep updates so it could be runnable on more recent versions of python) for now until i have more time to work on this again. 

glad to hear the fork is going strong :)